uli/basicengine-firmware #1: ESP32 port

antibyte, 2018-06-24:
Did you see Bitluni Labs work on generating color video in pure software on the EPS32 ?

[https://www.youtube.com/watch?v=-JXuwwXQh8c&t=22s](url)

Would it be possible to merge this and 

1) get cheaper as the expensive VS23S010 is no longer needed
2} have Bluetooth (Keyboard / Joystick)
3) have a dualcore CPU, more RAM and resources to have WIFI working
4) have more I/O

uli, 2018-06-25:
Thank you for your interest. I was not aware of this project, sounds interesting.

I'm supposed to tend to my day job right now, so I don't have time to look into this in detail. Two things,
though:
1. Savings from omitting the VS23 (and/or the I/O expander) would be partially (if not fully) eaten up by the more expensive ESP32.
2. If I understood correctly, the current implementation ties up on core completely and does not support concurrent use of WiFi. I don't know if Bluetooth would work.

That said, it would make for a system much easier to program; the majority of the development effort concerning the graphics for the BE went into squeezing as much data as possible through single SPI
and work around peculiarities in the block move feature. Simply having the data somewhere in RAM would simplify things tremendously (even though it would eat up a substantial part of the extra RAM in the ESP32), and would possibly allow for better (more/bigger/alpha-blended/whatnot) sprites/BGs as well.

I guess I'll have to get myself an ESP32 board and look at that demo...

antibyte, 2018-06-25:
The ESP32 is maybe 2-3$ more than the ESP8266
I ordered some VS23 from TME (and BE PCBs from China :) and they cost nearly the same as an ESP32 :( 
There is also the ESP-32 Wrover Module available, that adds 4 to 32 MB of SPI RAM while staying in the 5$ price range. 
http://www.electrodragon.com/product/esp32-wrover-v4-module-based-esp32/

BTW, posted the project to hackaday, they will have an article about it soon :)

uli, 2018-06-26:
OMG. :) Thanks for the heads-up; I used the opportunity to upload a new binary release.

More on-topic: I have taken a closer look at the ESP32 PAL code. There are a few more
restrictions, and they largely stem from the limitations of the DAC:

1. It's PAL only, and works because the maximum DAC sample rate is close to a multiple of the PAL color burst frequency. Bitluni has not succeeded in getting it to work with NTSC because of that. (My gut feeling is that that may be fixable with a bit of creative filtering of the output signal, but I don't know enough about the whole analog stuff to make a confident statement here. I do know that some displays have very low tolerances when it comes to color burst frequency, though...)
2. The pixel clock is limited by the DAC sample rate as well, and you will probably not be able to get more than the demo's 320 pixels per line, at least if you want color.

Anyway, ESP32 board's in the mail.

antibyte, 2018-06-27:
https://hackaday.com/2018/06/27/esp8266-home-computer-hides-unexpected-gems/

uli, 2018-06-28:
Thanks for spreading the word.

As for the ESP32 software video, it's not looking good, unfortunately. I have done extensive testing, have implemented NTSC as well as hybrid modes (PAL3.57 and NTSC4.43). I have also fixed some timings and levels that I believe to be incorrect, and have done a lot of wiggling of timings, frequencies, phase shift etc, but compatibility remains very low.

Daewoo LCD TV:
- PAL, PAL3.57, NTSC4.43: works
- NTSC: no color

Toshiba LCD TV:
- NTSC4.43: works (haven't had time to test other modes)

Samsung LCD TV:
- PAL: heavy color flickering
- NTSC: colored stripes, no color in some areas
- PAL3.57, NTSC4.43: invalid signal

Sony KX-14CP1 CRT monitor:
- no color in any mode

(All of these displays work 100% fine with both the BASIC Engine and a Raspberry Pi in both PAL and NTSC.)

That is simply not good enough, and I do not believe that there is much that can be done about it in software. The DAC clocks are fixed, so the phase shift/burst detection issues are pretty much unfixable without changing the main clock, which is probably going to break WiFi and/or Bluetooth.

I would be willing to accept patches that add support for the ESP32 and software video (as long as they don't break anything else), but I'm not likely to put any effort into this myself unless someone comes up with a solution that fixes at least the majority of these issues.

closed 2018-06-28 by uli