uli/basicengine-firmware #2: Failed to init synth

antibyte, 2018-06-30:
Just finished building my Basic Engine board, so i tried the examples.
In Shmup as soon as a sound should play i get "Failed to init synth in 1340"
Soundfont is on the sd card.

Sokoban has some strange graphics issues, but i think its WIP ?

And how do i get back to the root directory ?
Would be nice to add a hint to the documentation.

uli, 2018-07-02:
Wow, that was fast. :)

I think the problem is that I have never tried running shmup from a subdirectory; the sound font is loaded from a relative path. Could you try SOUND FONT "/sd/1mgm.sf2" and see if it works then?

[Sokoban exposes a CALL bug (see commit 9beca8e1c57), and needs fixes for the text window handling, which are also in the master branch since Saturday or something.]

antibyte, 2018-07-02:
Yes, SOUND FONT "/sd/1mgm.sf2" makes it work.

uli, 2018-07-12:
Fixed in v0.87-alpha.

closed 2018-07-12 by uli