uli/basicengine-firmware #7: PLAY command

bitartrate, 2019-07-07:
Hello Uli,
I have been working with the PLAY command for a couple of days and discovered there seems to be a bug when trying to select instruments above 9 in an SF2 file.  I am using firmware version v0.88-alpha-346.  I'm using PLAY "@9 C4" which seems to each match up with INST$(num) until 9.  If I PLAY "@10 C4" it plays instrument number 1.  I have found that if I start using letters A-Z I do get more instruments but I don't think they match up as if A = 10.  I think it is more like A=17.  Also, what happens after Z?
I really appreciate your work on the Basic Engine.  Thank you.

uli, 2019-07-11:
Please verify that this works in the latest nightly build and, if so, close the bug.

bitartrate, 2019-07-11:
Thank you, Uli. It does work as intended.

closed 2019-07-11 by bitartrate