uli/basicengine-firmware #10: need freespace error check

bitartrate, 2019-07-12:
Good morning/afternoon, Uli.  I have identified a bug when the SD card does not have enough memory to save a file.  The scenario is copying a file from /FLASH to /SD.  When /SD does not have enough memory the BASIC Engine freezes.  Suggest to implement a freespace check before allowing file copy.  Thank you.

uli, 2019-07-17:
Doing a free-space check is probably a good idea (and, generally speaking, how much space is left is something that the user should be able to find out), but writes to a full SD card should really not hang the system. 

There is nothing in sdfiles::fcopy() that should be able to block; I therefore assume that (our version of) SdFat doesn't handle disk-full conditions well. I have the slight suspicion that this might have something to do with it:
https://github.com/greiman/SdFat/commit/f0c65da00060fcc8c4ac9667d29b2d46557de42e

I have briefly tested the most recent version of SdFat including this fix, and it doesn't seem to break anything, so I have merged it into master. Please check if the next nightly build fixes this issue.

bitartrate, 2019-07-18:
This update to: basicengine-firmware-default-v0.88-alpha-354-g8985.tar.gz 
fixes the problem of the machine freezing if a file copy is performed to a full SD card.  It returns verbosely with "File write error".  Successful update.  Thank you, Uli!

closed 2019-07-18 by bitartrate