uli/basicengine-firmware #14: EDIT command Issues

jebielak, 2019-08-16:
The EDIT command editor lags badly after only a few lines.  I entered DATA lines, Example:
DATA 128, "GO"
DATA 129, "FOR"
...
DATA 143, "REM"

By the 27th line,  typing in text became exceedingly slow responding.  Also, back spacing didn't alway clear characters as it moved.

Tryed updating to the latest nightly and it was the same.

uli, 2019-08-19:
The slowdown is due to syntax highlighting, and the way the editor redraws the whole screen every time. The leftover deleted characters are an unrelated issue. Both should be fixed in the next nightly build; please check if that works for you.

uli, 2019-08-19:
Oh, and that code you are writing there does have something to do with Commodore BASIC, doesn't it? :)

jebielak, 2019-08-19:
Yes, I'm writing a Commodore Basic to Basic Engine converter.  I decided to put the tokens in DATA statements in the initial build, and once I've got the the detokenizing and alternate functions working, such as POKE, PEEK, and the PETSCI printing, for the C64 , moving them to separate files so It can convert Commodore Basis V2 through V10, and then eventually TRS, Apple, PC BASICA, Etc.  I wrote the same program back in the 80's, and a variant that converted to Pascal.

uli, 2019-08-19:
Very nice.

I have found out, though, that the fix I pushed earlier does not actually fix the backspace issue, so I'm assigning this bug back to me.

uli, 2021-10-21:
Probably not going to be fixed in OG builds, and NG has a completely new editor.