uli/basicengine-firmware #15: Is it real to add VGA output for esp32 version?

fbezruchko, 2019-08-29:
Its not an issue. But is it real to add VGA output instead composite for esp32 version? Can FabGL Lib helps for this?

uli, 2019-08-30:
I just had  a quick look at it, and, yes, I think that would be possible; it seems the VGA output can be configured to use any available GPIOs, and the library doesn't seem to use any weird tricks like the PAL generator.
(Also, though not related to video output, I like how it uses the ULP to talk to PS/2 devices...)

That said, I'm not likely to find the time to implement this any time soon. I'll be happy to accept any patches, though.

fbezruchko, 2019-08-30:
I hope can find enougth time for this. I ordered wroom-32 need some time for waiting. What is the main module of video output in basic engine source? It seems i need add additional #ifdef in video.h and implement new something like "vgagfx" module (or class, *.h+*.cpp). Is there anything else needed to patch?

OzHawk, 2020-07-30:
Has this enhancement been advanced at all?
I'm looking for a BASIC interpreter for use on an ESP32 with VGA output.  I wanted something more than tiny basic.

uli, 2021-03-08:
I'm going to concentrate on the Allwinner H3 platform as an upgrade from the original BASIC Engine because it's way superior to ESP32-based devices (such as the VGA32) at almost the same price point. I therefore don't think that this is going to happen after all. Sorry.

closed 2021-03-08 by uli