uli/basicengine-firmware #18: SAVE PCX command not respecting POS and SIZE parameters and always errors but still saves image file

bitartrate, 2019-09-07:
Hello Uli,
From handbook: SAVE PCX image$ [POS x, y] [SIZE w, h]

Whether executed from user space or within a program SAVE PCX ignores parameters and copies the whole screen.  It errors with "Syntax error (exp command)" but still saves the image file.

Also, the command does not follow Engine BASIC conventional syntax.  It will work as:
SAVE PCX "image.pcx",POS x,y SIZE w,h      or
SAVE PCX"image.pcx",x,y,w,h

But not the conventional: SAVE PCX "image.pcx" POS x,y SIZE w,h

Thank you!

konimaru, 2019-09-09:
The syntax error happens because command _consumption_ stops at the first comma after the filename. IOW the save command is simply done without POS/SIZE. The issue will be sorted with pull request #20 (or for now from my fork).

bitartrate, 2019-09-09:
Great job Marko!  I have compiled your fork and verified that SAVE PCX now works as intended.  Thank you.

uli, 2019-09-16:
Please do not close bugs until they have been fixed and tested in _this_ repository.

uli, 2019-10-07:
Does this work now?

bitartrate, 2019-10-07:
This does work as expected.  Closing topic. Thank you.

closed 2019-10-07 by bitartrate