uli/basicengine-firmware #21: RENUM does not check for RESTORE command and change the line number

bitartrate, 2019-09-15:
When using RENUM does not check for RESTORE commands to change the line number it points to.  Would like it to function as it does for GOTO commands.

konimaru, 2019-09-15:
Fixed in my branch despite having messed up the commit comment. It's still early here :)

bitartrate, 2019-09-15:
I saw that you had made that commit right before I went to bed.  I knew what you meant ;)
I have verified that your fix to RENUM does interact with RESTORE correctly now.  Thank you!

uli, 2019-09-16:
Please do not close bugs until they have been fixed and tested in _this_ repository.

bitartrate, 2019-09-16:
Sorry about that Uli.  I'm understanding the workflow slowly but surely.

uli, 2019-09-30:
FIxed in commit ee38433d5b074dc01200ef0ef4a74633959c79cf. Please close if it works now.

bitartrate, 2019-09-30:
I have compiled today and verified it works as intended.  Thank you.

closed 2019-09-30 by bitartrate