uli/basicengine-firmware #26: Feature request to look for and run ".autoexec" when first powered on

bitartrate, 2019-09-19:
I have several projects that would be greatly enhanced if there where an ".autoexec" type functionality.  What I propose is giving the BASIC Engine the ability to look for and run this specially named basic program on startup.

konimaru, 2019-09-23:
I have a working prototype for this. Right now it attempts to execute the command ```run ".autoexec"``` provided the relevant config option (9) is enabled. It doesn't enforce a specific path but uses the current directory for file access.

Another option would be to always load a stub from **/flash** (i.e. ```run "/flash/.autoexec"```) which can then do 2nd level loading.

Thoughts?

uli, 2019-09-23:
Sounds good to me. Two things, though: 
1. In my not so humble opinion, the file should be called "AUTOEXEC.BAS" (no dot).
2. There has to be a reliable way to prevent it from being executed if something goes wrong, especially if it resides in /flash.

konimaru, 2019-09-23:
1. And always in uppercase, right?
2. Now I know why I had a bad feeling about **/flash**. In this case I would stick with **cwd**. That said, can **cwd** ever be **/flash** or will it always be **/sd** or none?

konimaru, 2019-09-23:
A quick fix would be ```run "/sd/AUTOEXEC.BAS"``` ...

bitartrate, 2019-09-23:
I think you two are on a good track with this.  I agree with AUTOEXEC.BAS as the filename.  Possibly look in /flash first and then /sd second.  I don't think it needs to enforce a path as at startup BE always is in the root of /sd.

konimaru, 2019-09-23:
To keep it simple I just went back to using **CWD$()** as the source of AUTOEXEC.BAS. If no SD card is mounted then it attempts to find it in **/flash**.

**IF** something goes wrong with the **/flash** version one can either insert an SD version (clean/working) or ground PSX_DATA during poweron which skips the autostart feature entirely.

[autostart branch](https://github.com/konimaru/basicengine-firmware/tree/autostart)

bitartrate, 2019-09-24:
Hi Konimaru, I did clone your autostart branch but it came in without the changes.  So I copied the changed files by hand.  I was just about to say I couldn't get it to load AUTOEXEC.BAS but I just remembered CONFIG 9.  So far so good it is working for me from both /sd and /flash.  Good job!

konimaru, 2019-09-24:
What command did you use (for cloning)?

konimaru, 2019-09-24:
What command did you use (for cloning)?

git clone **--branch autostart** http://github.com/konimaru/basicengine-firmware.git

bitartrate, 2019-09-24:
ahhh I see, no I simply used 'git-clone https://github.com/konimaru/basicengine-firmware.git'
Thank you.  That adds to my understanding of Git.

konimaru, 2019-09-25:
Back to square one. vs23 is not available on all platforms.

konimaru, 2019-09-25:
@uli would a dummy **digitalRead** on other platforms (e.g. esp32) be an option or do you prefer a keyboard/pad solution for the override?

konimaru, 2019-09-25:
@bitartrate: Could someone who has a PSX controller available please check my **autostart** branch?

When auto start is enabled (config 9,1) pressing any button affecting pad(0) should cancel auto start.

bitartrate, 2019-09-25:
@konimaru I confirm that the PSX controller does cancel the autostart with message "autostart cancelled"
Excellent!

konimaru, 2019-09-25:
Thanks for testing this.

bitartrate, 2019-09-25:
You are very welcome.  Thank you for the feature.

bitartrate, 2019-09-25:
Are you contemplating loading from /FLASH ?

konimaru, 2019-09-25:
The way it's done now is that I inject a small basic command(!) into the buffer and pass it straight on to the tokenizer. I'm not sure the needed functionality can be placed in a single line. Feel free to surprise me.

bitartrate, 2019-09-25:
I might just do that :)

peek82, 2021-02-08:
Did this ever get merged into the firmware? I saw there were commits & pull requests by @konimaru but on my machine CONFIG 9,# applies/reverts the recent code formatting change, and I'm not seeing a CONFIG 10,#.

konimaru, 2021-02-08:
> Did this ever get merged into the firmware?

Not to my knowledge. Since it was never merged on time, the pull requests quickly _collected_ merge conflicts which I only fixed a couple of times ...

uli, 2021-02-08:
So this is why nobody should elect me as President of the Universe: I didn't merge this because I consider the implementation way too intrusive for such a small feature. I didn't say anything, though, because I don't like to say things like that, and I didn't write a better solution either. Sorry about that. :(

Anyway, I wrote something simpler now. I left out the config part, though, because I feel it's redundant. After all, you can disable the autoexec temporarily by holding a key, and permanently by deleting the file.

peek82, 2021-02-10:
I can confirm this is working on my machine. I have the AUTOEXEC.BAS retrieving the time from a Maxim DS3231 RTC and setting the time on the Basic Engine. Thanks guys.

uli, 2021-03-08:
Thanks for testing.

closed 2021-03-08 by uli