uli/basicengine-firmware #27: sdfiles::rmdir() doesn't work as expected

konimaru, 2019-09-23:
**mkdir "some_name"** creates a directory in the CWD. Removing said directory with **rmdir "some_directory"** does not work (I get font corruption which is cured by calling e.g. **files** again).

The directory can in fact only be removed with **rmdir "/sd/some_directory"**.

konimaru, 2019-09-24:
Font corruption is most likely due erroring out after SD_BEGIN() without matching SD_END().

konimaru, 2019-09-24:
Fixed locally (pending verification).

bitartrate, 2019-09-24:
In the official nightly build I have verified the font corruption but rmdir "/sd/some_directory" did not remove the directory for me.
I cloned your repo with git clone --branch rmdir http://github.com/konimaru/basicengine-firmware.git
But I still see the font corruption with rmdir"some_directory"
With this build the rmdir"/sd/some_directory" did remove the directory though.

konimaru, 2019-09-24:
Well that would be because I haven't committed any changes yet :)

bitartrate, 2019-09-24:
You did say 'locally', didn't you :)  ::palm face::

bitartrate, 2019-09-24:
I have confirmed that rmdir does work as intended by this fix.

PR #33 (2019-09-25): sdfiles: rmdir()/mkdir(): fixed relative path removal and SPIFFS filter
Fixes #27.
closed 2019-09-25 by uli