uli/basicengine-firmware #37: RECT command is inconsistant

konimaru, 2019-09-27:
fill colour == -1
- rectangle is one pixel short in each direction, e.g.
  - ```RECT100,100,102,102,200,-1``` only gives me 2x2 pixels

fill color != -1
- x dimension is now correct but height is still one short (3 by 2 with the example above)
- fill colour is in fact line colour
- I'd expect the border still to be drawn in line colour, this isn't the case
- special cases **w=0 || h=0** and **w=0 && h=0** are not handled

bitartrate, 2019-09-27:
I thought I was seeing this but I was always drawing big rectangles and was not sure I could trust my eyes.  I have confirmed your bug.

uli, 2019-10-04:
I'll assume this works now.

closed 2019-10-04 by uli

konimaru, 2019-10-11:
Works for me. Ta