uli/basicengine-firmware #38: Unable to compile for Allwinner H3

bitartrate, 2019-09-28:
Hi, Uli,

I have become interested in compiling for the OrangePi H3.  The makefile halts with error:
"Makefile.h3:19: /home/kevin/devel/orangepi/allwinner-bare-metal/build.mk: No such file or directory"
The closest I have come to getting past this error was to make an empty file 'build.mk'
then add '#INCLUDE = ~/devel/orangepi/allwinner-bare-metal/build.mk'
to the makefile.  It satifies the missing file but ends without making a .bin file; at least not that I can find.

I do believe I have the necessary H3 toolchain installed.

uli, 2019-09-30:
build.mk is part of the allwinner-bare-metal git repo. It's in the "opi" branch, though. Maybe that's the problem...

bitartrate, 2019-09-30:
That was half the problem.  Thank you.
Now I see I need to install crosstools-NG? as allwinner-bare-metal is looking for folder 'x-tools'.

uli, 2019-10-02:
That is correct. The configuration file for the toolchain is `crosstool-ng.config`.

bitartrate, 2019-10-02:
Unfortunately I am not finding the correct documentation for x-tool.  I have crosstool-ng-1.24.0 installed but I am not able to find the docs to put that together with your allwinner-bare-metal opi branch.

uli, 2019-10-04:
````
mkdir xyz
cd xyz
cp ../allwinner-bare-metal/crosstool-ng.config .config
ct-ng build
````

bitartrate, 2019-10-06:
I am making headway.  My latest effort looked like it was going to work but after the lengthy build log it ended in these errors:

In file included from ttbasic/video.h:13,
                 from ttbasic/mcurses-config.h:18,
                 from ttbasic/mcurses.cpp:18:
h3/h3gfx.h: In member function 'void H3GFX::setPixels(uint32_t, pixel_t*, uint32_t)':
h3/h3gfx.h:84:5: error: 'memcpy' was not declared in this scope
     memcpy((pixel_t *)address, data, len * sizeof(pixel_t));
     ^~~~~~
h3/h3gfx.h:84:5: note: 'memcpy' is defined in header '<cstring>'; did you forget to '#include <cstring>'?
h3/h3gfx.h:10:1:
+#include <cstring>
 
h3/h3gfx.h:84:5:
     memcpy((pixel_t *)address, data, len * sizeof(pixel_t));
     ^~~~~~
<builtin>: recipe for target 'ttbasic/mcurses.o' failed
make[1]: *** [ttbasic/mcurses.o] Error 1
make[1]: Leaving directory '/media/kevin/500gbssdpart2/OpenSourceComputers/BasicEngineComputer/Firmware/basicengine-firmware'
Makefile:31: recipe for target 'h3' failed

bitartrate, 2019-10-07:
I wish I had more experience...
We are past the previous error.  Now make ends with:
make -f Makefile.h3
make[1]: Entering directory '/media/kevin/500gbssdpart2/OpenSourceComputers/BasicEngineComputer/Firmware/basicengine-firmware'
make[1]: *** No rule to make target '/home/kevin/devel/orangepi/allwinner-bare-metal/libos.a', needed by 'basic.elf'.  Stop.
make[1]: Leaving directory '/media/kevin/500gbssdpart2/OpenSourceComputers/BasicEngineComputer/Firmware/basicengine-firmware'
Makefile:31: recipe for target 'h3' failed
make: *** [h3] Error 2

I see that libos.a and linker.ld are referenced in build.mk but I do not see where these are built.

Thank you.

c0d3Art1st, 2021-08-28:
Hey Uli,
unfortunatelly I'm also not able to the get the firmware compiled for the h3. I followed the short instructions on basicengine.org and your hints in this issue, but I still get the following error, when when using: `ninja -f build.ninja.h3`

ninja: error: '.git/index', needed by 'ttbasic/version.h', missing and no known rule to make it

Any clue, what the problem might be. Help would be greatly appreciated :-)

Thanks a lot

uli, 2021-08-29:
I suppose you downloaded the repo as a zip file or something like that. Use git to check out the source tree, and you will have a .git directory with an index file in it.
(I'll close this because it's not a bug. Please ask any follow-up questions in the forum. Thank you.)

closed 2021-08-29 by uli