uli/basicengine-firmware #40: SDL version SOUND bug

bitartrate, 2019-09-28:
When I type 'SOUND 0,1,10,10' at the start of a session sound works in both immediate mode and in a one line program.
But once I load a lengthy basic program into memory it returns error "Failed to init synth" in both
program mode and immediate mode.

uli, 2019-09-30:
The SDL build reads the sound font from the current directory, unless the environment variable ENGINEBASIC_ROOT specifies a different (absolute) path. The reason is that I'm not quite sure what an appropriate directory structure would look like, since the SDL build can run on pretty much anything.

(My motivation for making that port in the first place is to make an Antix Linux Live CD that can be used to boot into Engine BASIC on old (ca. 1998 and up) PCs.)

I guess a reasonable solution would be to have a command-line parameter for setting the root directory, and to default to the cwd if none is specified.

bitartrate, 2019-09-30:
After I realized that it looks in the CWD it became clear why and I do like that functionality.  I am not sure I would recommend anything different at the moment.  Maybe in time after I make a dedicated Linux install a workflow will emerge.  I REALLY like that you made an SDL version.  I have run it in full screen and it is really great.

uli, 2019-10-04:
The SDL build now defaults to CWD as root, and there is also the "-r" command line option, so this should not be an issue anymore.

closed 2019-10-04 by uli