uli/basicengine-firmware #41: SDL version pixel graphics color bug

bitartrate, 2019-09-29:
This affects LINE, CIRCLE, RECT, and PSET commands.  The color parameter only outputs a shade of blue no matter what value is given between 0-255.  The same bug exists in palette 0 and 1.  
However the COLOR and GPRINT work as intended.

bitartrate, 2019-09-29:
If you fill the screen with random lines and colors then SAVE PCX the resulting .pcx file does save the colors as can be seen in a paint program.

uli, 2019-09-30:
Please check if that fixes the issue.

Also, please open another issue for the SAVE PCX bug, it's unrelated.

bitartrate, 2019-09-30:
What works:  color, the border and fill color of both CIRCLE and RECT work.

Minor issue:
CIRCLE: if you choose a radius that will draw the circle outside the screen boundary SDL closes BE window.
Good Job!

bitartrate, 2019-09-30:
I'm sorry the SAVE PCX was informational not a bug report.

uli, 2019-10-04:
The crash should be fixed now, so I will close this.

closed 2019-10-04 by uli