uli/basicengine-firmware #47: SDL fullscreen (switch '-f') needs a way to exit back to host OS desktop

bitartrate, 2019-10-02:
When in fullscreen mode in the SDL BASIC Engine there is no way to cleanly close out of it back to the host OS desktop.  Something like an immediate mode EXIT command.

bitartrate, 2019-10-04:
I like the SYSTEM command.  This is kind of a little thing but would it be possible to make SYSTEM reduce back to a window?  My use case would be to have Engine BASIC documentation loaded on the host and be able to go back and forth from fullscreen to refer to it without closing BE/SDL.

uli, 2019-10-07:
I think a keyboard shortcut is more useful for that.

uli, 2019-10-11:
So does Alt-Return resolve your issue then?

bitartrate, 2019-10-12:
Uli, that is perfect!  Thank you.
To recap we now have SYSTEM command to close the SDL down.
<alt-return> to switch between fullscreen and windowed mode.

closed 2019-10-12 by bitartrate