uli/basicengine-firmware #48: SDL version crashes when using SAVE CONFIG

bitartrate, 2019-10-02:
This seems like it will end up being related to issue #40 
SAVE CONFIG crashes the SDL version.

bitartrate, 2019-10-02:
The native version when using CHDIR crashes/locks-up the BE making an "engine" type sound.
Could be related to this issue #48

uli, 2019-10-03:
No, that was related to the syntax change. Just pushed a fix.

bitartrate, 2019-10-03:
As of firmware ver.548.  Native version no longer crashing due to above syntax change

bitartrate, 2019-10-03:
On the SDL side SAVE CONFIG now makes a .config file but does not seem to read it when restarting basic.sdl

uli, 2019-10-04:
Can't reproduce that, works for me.

bitartrate, 2019-10-04:
I have expanded my config settings testing and find that some config settings do get saved and work such as beeper volume and font.  It is the SCREEN mode that I cannot get to come up on the next launch.

uli, 2019-10-07:
That should take care of it.

Does this issue also exist on ESP8266?

bitartrate, 2019-10-07:
firmware v.567 on the SDL side is now correctly saving and recalling CONFIG 4,x

however v.567 in native is not saving/recalling CONFIG 4,X.  I did test it with CONFIG 7,X and that is working.

konimaru, 2019-10-11:
Saving the mode is OK but after the config is loaded there is never a **vs23.setMode()** call done and **vs23.being** doesn't take a mode parameter.

uli, 2019-10-11:
That should do. Please give me feedback on how it works, I have not actually tested these platforms yet.

konimaru, 2019-10-11:
Works for **hosted**.

bitartrate, 2019-10-12:
Works for SDL and native versions