uli/basicengine-firmware #54: Coordinate clamping not working with LINE and PSET.  Inter related; screen buffer also does not clear correctly.

bitartrate, 2019-10-08:
The LINE command and the PSET commands are not clamping. If one plots outside the resolution area both the LINE and PSET command do not clamp.  Upon exiting a program or immediate mode the screen buffer does not clear out which causes the pixels in memory below the bottom of the screen mode to scroll up indefinitely while editing.  You must change to a different screen mode than the one you are in to clear it out.

 CIRCLE command clamps.  RECT command is good; it will give error code if illegal coords are input.

konimaru, 2019-10-09:
None of the drawing commands are clipped to the visible area. It's always **0..psize(2)-1**.

uli, 2019-10-09:
It is. The catch is that there is (half a) blank line in video memory that is used to accelerate scrolling the text screen. That line is right below the visible screen and thus easily gets overwritten by commands drawing across the bottom of the screen. The same goes for a section of video memory that is used as the color memory for text mode.

A semi-solution would be to move the areas used by the system to the very end of video memory, by changing the bin allocator strategy. Another would be to introduce a user-definable graphics clipping window.

konimaru, 2019-10-09:
> The catch is that there is (half a) blank line in video memory that is used to accelerate scrolling the text screen.

I'm just about done eliminating that.

uli, 2019-10-09:
Could you give me a heads up of what files you are currently working on? I intend to reformat much of the source code using clang-format in an effort to make it slightly less unreadable, and to save me the work of actually writing a coding style. To avoid conflicts I would want to start with the stuff nobody is touching ATM.

konimaru, 2019-10-09:
Just **tvutil.cpp** but I can wait. Needs some real live testing on the h/w. I see benefits on **hosted** in that there is more off-screen space available. But I still need some clarity on blitter effects.

uli, 2021-03-08:
The blanking line is gone, so this should not be a problem any longer.

closed 2021-03-08 by uli