uli/basicengine-firmware #59: SDL version will no longer make

bitartrate, 2019-10-10:
Today's repo build for SDL version ends in error:
/usr/bin/ld: libraries/lua/lapi.o: Relocations in generic ELF (EM: 40)
/usr/bin/ld: libraries/lua/lapi.o: Relocations in generic ELF (EM: 40)
/usr/bin/ld: libraries/lua/lapi.o: Relocations in generic ELF (EM: 40)
/usr/bin/ld: libraries/lua/lapi.o: Relocations in generic ELF (EM: 40)
/usr/bin/ld: libraries/lua/lapi.o: Relocations in generic ELF (EM: 40)
/usr/bin/ld: libraries/lua/lapi.o: Relocations in generic ELF (EM: 40)
libraries/lua/lapi.o: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
Makefile.sdl:32: recipe for target 'basic.sdl' failed
make[1]: *** [basic.sdl] Error 1
make[1]: Leaving directory '/media/kevin/500gbssdpart2/OpenSourceComputers/BasicEngineComputer/Firmware/basicengine-firmware'
Makefile:35: recipe for target 'sdl' failed
make: *** [sdl] Error 2

uli, 2019-10-11:
I presume you compiled the H3 build before?
The object files for all non-Arduino builds (SDL, DOS, H3) are all stored in the same place, so if you have a previous build for a different platform, linking will fail for obvious reasons.

Basically, the Makefiles need a fix to store object files in different directories.

bitartrate, 2019-10-12:
I understand now.