uli/basicengine-firmware #67: PEEK/POKE broken? or what number base form does it expect?

bitartrate, 2019-11-15:
I tried decimal numbers, hex, binary, and with all the usual ending modifiers.  Need to know what base and form PEEK/POKE expects.  All I get are unmapped address errors.   Even the timer config register 60000600h comes up unmapped.

uli, 2019-11-17:
Hmmm... that should actually work. The base is 0. I have tried "POKE $60000600, 0" on the hosted build, and it looks fine (i.e. segfaults as expected).

I don't have an ESP8266 system at hand right now, so I can't test it there.

bitartrate, 2019-11-19:
Hi Uli, are you saying that you need an ESP8266 system?  Or just didn't have one where you where at the moment?

bitartrate, 2019-11-21:
Closing this issue as these commands are not broken.  To sum up for others who may see this with an example: "POKE $40000080,1" is a reset from the ESP8266 memory map and that is the form these commands expect.
Thank you.

closed 2019-11-21 by bitartrate