uli/basicengine-firmware #68: WAIT command will not wait when end time overflows

konimaru, 2019-11-22:
current code:
```
void Basic::iwait() {
  int32_t tm;
  if (getParam(tm, 0, INT32_MAX, I_NONE))
    return;
  uint32_t end = tm + millis();
  while (millis() < end) {
    process_events();
    uint16_t c = sc0.peekKey();
    if (process_hotkeys(c)) {
      break;
    }
    yield();
  }
}
```
Example: **millis()** returns close to 0xFFFFFFFF, **end** time == 0x0xxxxxxx
- due to this being **uint32_t** the **while** will never even try to wait
- treating the comparison as signed would solve the problem here (bit 31: 1 to 0) but re-introduces it at (bit 31: 0 to 1)

Suggested solution (with the limitation that the max delay is limited to 2^31 ms):
```
void Basic::iwait() {
  int32_t tm;
  if (getParam(tm, 0, INT32_MAX, I_NONE))
    return;
  uint32_t end = tm + millis();
  while ((signed)(millis()-end) < 0) {
    process_events();
    uint16_t c = sc0.peekKey();
    if (process_hotkeys(c)) {
      break;
    }
    yield();
  }
}
```

peek82, 2021-02-08:
@konimaru 
Once in a while my machine freezes and when I hit ctrl-c to break, it's always on a line with a WAIT command. I had wondered if there was an issue with the rollover in WAIT...  I wonder if my issue is related to this bug?

uli, 2021-03-08:
I have now merged the overflow fix into Arduino_wifi's sdknowifi tree, and it seems that that fixes the issue with the WAIT command. It should show up in the next git build. (Note, however, that the next build will only be triggered when there is a change in the basicengine-firmware repo.)

closed 2021-03-08 by uli