uli/basicengine-firmware #70: SDL: MoveBlock shows different overlap behaviour compared to hosted/native

konimaru, 2019-11-29:
SDL's MoveBlock function doesn't work **vs23s010** style. The latter honors live updates while moving stuff around. SDL simply takes a snapshot of what is there and copies it somewhere else.

I'd argue all platforms should have the same behaviour, in this case following h/w.

see #58

uli, 2019-11-30:
Hmmm... I think replicating the exact behavior of the VS23 block move is not a winning move. It's complicated and hurts performance.

Since this arose from the issue of having to fill an area with a solid color quickly, this is the approach I would favor:

1. Make MoveBlock() internal to the VS23 driver,
2. add a generic interface for blitting (copyArea() or something like that) the behavior of which is undefined if the source and destination overlap,
3. add a method fillArea() with a generic implementation using drawLine(),
4. add optimized implementations of fillArea() for VS23 (using your code) and SDL (using SDL_FillRect()).
5. use fillArea() to clear text lines.

I'll look into that.

uli, 2019-12-01:
Done. If you could, please have a look at 2e394da18f1a9c92bd5ea65eafca30fde33a2035 and see if there is anything that doesn't look right. We probably need to add a condition that falls back to Video::fillRect() for cases that are not supported.

konimaru, 2019-12-02:
Looks OK so far. **hosted** doesn't show anything abnormal, I'll run it on my boards later.

konimaru, 2019-12-03:
Works for me. Thanks.

closed 2019-12-03 by konimaru