uli/basicengine-firmware #71: make interpreter skip over PROC or subroutines defined by &label:

bitartrate, 2019-11-30:
Would it be reasonable to make the interpreter ignore a non-called PROC or subroutine defined with a  "&label:" if encountered?  It could add more programmer flexibility and remove under-flow errors while making Engine BASIC that much more dynamic and user friendly.  RETURN of course would flag the interpreter where to resume.

uli, 2019-12-01:
The problem is how to identify with any degree of certainty where a procedure ends. Engine BASIC does not have anything resembling code blocks, making finding the end of a procedure effectively complete guesswork: RETURN can occur multiple times in one procedure, and not even necessarily at the end if you throw in a few strategic GOTOs.

A possible solution would be to introduce and optional end-of-procedure marker ("END PROC") that can be used to determine the end of a procedure with confidence. If such a marker is not present, it would effectively fall back to the current behavior.

Any thoughts?

bitartrate, 2019-12-01:
This is why you da man Uli!  I tend to want to think of procedures as programmer defined commands.  Toward that way of thinking it seems to make the most sense to give this functionality to procedures over subroutines; however, if it is simple enough I wouldn't want to leave subroutines out of this functionality.  I agree with you that "END PROC" seems to be a good way to go.