uli/basicengine-firmware #74: CHAR command formatting

JamesCWhite, 2020-12-18:
When using the CHAR command to print a character to the screen as described here: https://basicengine.org/manual.html#_char
The code is formatted to remove the space, so typing "CHAR X,Y,C" works and after a LIST it strips the space, showing "CHARX,Y,C". At that point the code still runs while still in memory, but if loaded from a saved file the CHAR command is not recognized, leading to an 'Unknown Command in # (CHARX)'

For a workaround it is possible to instead use LOCATE X,Y to position the cursor followed by a PRINT CHR$(C)

uli, 2020-12-18:
That should do it. I hope the build service still works, haven't committed anything in a while...

JamesCWhite, 2020-12-18:
Thank you Uli!!! Your update fixed the bug and it compiled just fine. I really appreciate everything you have done on this project, rock on man!

closed 2020-12-18 by JamesCWhite