uli/basicengine-firmware #86: Some strings will not be translated even if you manually add them to the translation file

CityAceE, 2024-05-08:
Here are all untranslatable strings.

"ЁЮЖЙ" is just a marker for the translated string.

```
{
  "command": "BG",
  "tokens": [ "BG" ],
  "section": "bg",
  "type": "bc",
  "brief": "ЁЮЖЙ",
  "desc": "ЁЮЖЙ\nЁЮЖЙ",
  "usage": "BG bg [TILES w, h] [PATTERN px, py, pw] [SIZE tx, ty] [WINDOW wx, wy, ww, wh]\n      [PRIO priority] [<ON|OFF>]\nBG OFF",
  "args": {
    "bg": "background number [\\Fn0\\Ff to \\Fn{MAX_BG_m1}\\Ff]",
    "w": "background map width, in tiles",
    "h": "background map height, in tiles",
    "px": "tile set's X coordinate, pixels [\\Fn0\\Ff to \\FnPSIZE(0)-1\\Ff]",
    "py": "tile set's Y coordinate, pixels [\\Fn0\\Ff to \\FnPSIZE(2)-1\\Ff]",
    "tx": "tile width, pixels [\\Fn8\\Ff (default) to \\Fn255\\Ff]",
    "ty": "tile height, pixels [\\Fn8\\Ff (default) to \\Fn255\\Ff]",
    "wx": "window X coordinate, pixels [\\Fn0\\Ff (default) to \\FnPSIZE(0)-1\\Ff]",
    "wy": "window Y coordinate, pixels [\\Fn0\\Ff (default) to \\FnPSIZE(1)-1\\Ff]",
    "ww": "window width, pixels [\\Fn9\\Ff to \\FnPSIZE(0)-wx\\Ff (default)]",
    "wh": "window height, pixels [\\Fn0\\Ff to \\FnPSIZE(1)-wy\\Ff (default)]",
    "priority": "Background priority [\\Fn0\\Ff to \\Fn{MAX_BG_m1}\\Ff, default: \\Fnbg\\Ff]"
  },
  "note": "ЁЮЖЙ\nЁЮЖЙ",
  "bugs": "ЁЮЖЙ",
  "ref": [ "LOAD BG", "LOAD PCX", "MOVE BG", "SAVE BG" ]
},

{
  "command": "COMPARE",
  "tokens": [ "COMPARE" ],
  "section": "fs",
  "type": "bf",
  "brief": "ЁЮЖЙ",
  "usage": "result = COMPARE(file1$, file2$)",
  "args": {
    "file1$": "name of first file to compare",
    "file2$": "name of second file to compare"
  },
  "ret": "ЁЮЖЙ\nЁЮЖЙ"
},

{
  "command": "DO",
  "tokens": [ "DO" ],
  "section": "bas",
  "type": "bc",
  "brief": "ЁЮЖЙ",
  "usage": "DO\n  instruction_block\nLOOP [<WHILE|UNTIL> condition]",
  "args": {
    "instruction_block": "one or more instructions on one or more lines",
    "condition": "a numeric expression"
  },
  "note": "ЁЮЖЙ\nЁЮЖЙ",
  "example": "\\Fni=0\nPRINT \"Value of i at beginning of loop is \";i\nDO\n  i=i+1\nLOOP WHILE i<10\nPRINT \"Value of i at end of loop is \";i\n\\Ff"
},

{
  "command": "FN",
  "tokens": [ "FN" ],
  "section": "bas",
  "type": "bn",
  "brief": "ЁЮЖЙ\nЁЮЖЙ",
  "usage": "v = FN procedure[(argument[, argument ...])]",
  "args": {
    "procedure": "ЁЮЖЙ",
[... 162 lines cut ...]
  "brief": "ЁЮЖЙ",
  "usage": "SCREEN mode",
  "args": {
    "mode": "screen mode [\\Fn1\\Ff to \\Fn10\\Ff]// XXX: No convenient way to auto-update the number of modes."
  },
  "modes": "ЁЮЖЙ\n=== header\n| ЁЮЖЙ | ЁЮЖЙ | ЁЮЖЙ\n| 1 | ЁЮЖЙ | ЁЮЖЙ\n| 2 | ЁЮЖЙ | ЁЮЖЙ\n| 3 | ЁЮЖЙ |\n| 4 | ЁЮЖЙ | ЁЮЖЙ\n| 5 | ЁЮЖЙ | ЁЮЖЙ\n| 6 | ЁЮЖЙ | ЁЮЖЙ\n| 7 | ЁЮЖЙ | ЁЮЖЙ\n| 8 | ЁЮЖЙ | ЁЮЖЙ\n| 9 | ЁЮЖЙ | ЁЮЖЙ\n| 10 | ЁЮЖЙ | ЁЮЖЙ\n===",
  "note": "ЁЮЖЙ\nЁЮЖЙ\nЁЮЖЙ",
  "ref": [ "BG", "CLS", "FONT", "PALETTE", "SPRITE" ]
},

{
  "command": "VSYNC",
  "tokens": [ "VSYNC" ],
  "section": "scr",
  "type": "bc",
  "brief": "ЁЮЖЙ",
  "desc": "ЁЮЖЙ",
  "usage": "VSYNC [frame]",
  "args": {
    "frame": "ЁЮЖЙ"
  },
  "note": "Specifying a frame number helps avoid slowdowns by continuing immediately if the program is \"late\", and not wait for another frame. ==== ---- f = FRAME DO ' stuff that may take longer than expected VSYNC f+1\t// <1> f = f + 1 LOOP ---- <1> This command will not wait if frame \\Fnf+1\\Ff has already passed. ====",
  "ref": [ "FRAME" ]
},
```

uli, 2024-05-08:
That should do it without having to change all whitespace to tabs.

CityAceE, 2024-05-08:
There's still a couple of problems. I think because of quotes.

```
{
  "command": "GPOUT",
  "tokens": [ "GPOUT" ],
  "section": "io",
  "type": "bc",
  "brief": "ЁЮЖЙ",
  "usage": "GPOUT pin, value (ESP8266)\n\nGPOUT port, pin, value (H3)",
  "args": {
    "port": "ЁЮЖЙ",
    "pin": "ЁЮЖЙ",
    "value": "pin state [\\Fn0\\Ff for \"low\", anything else for \"high\"]"
  },
  "note": "ЁЮЖЙ\nЁЮЖЙ",
  "ref": [ "GPIN", "GPMODE" ]
},
```

```
{
  "command": "VSYNC",
  "tokens": [ "VSYNC" ],
  "section": "scr",
  "type": "bc",
  "brief": "ЁЮЖЙ",
  "desc": "ЁЮЖЙ",
  "usage": "VSYNC [frame]",
  "args": {
    "frame": "ЁЮЖЙ"
  },
  "note": "Specifying a frame number helps avoid slowdowns by continuing immediately if the program is \"late\", and not wait for another frame. ==== ---- f = FRAME DO ' stuff that may take longer than expected VSYNC f+1\t// <1> f = f + 1 LOOP ---- <1> This command will not wait if frame \\Fnf+1\\Ff has already passed. ====",
  "ref": [ "FRAME" ]
},
```