uli/basicengine-firmware #116: Bpim2z-lt: system reboots

CityAceE, 2024-05-29:
basicengine-bpim2z-lt-2024-05-28

Steps to reproduce:

1. Input `system'
2. Press Enter

Another way:

1. Input `shell`
2. Press Enter
3. Input `mc`
4. Press Enter

Free mem reduced after every reboot.

There are no any messages in UART.

uli, 2024-05-29:
> Steps to reproduce:
> 
>     Input system
>     Press Enter
> 
That's (sort of) intentional. `SYSTEM` exits the Engine BASIC interpreter. On the embedded platforms, the BASIC interpreter is respawned when it exits, so `SYSTEM` effectively reboots the interpreter. Might want to document that somewhere...

> Another way:
> 
>     Input shell
>     Press Enter
>     Input mc
>     Press Enter

That should not happen; sounds like a crash. Does this happen on H3 LT systems as well?

> Free mem reduced after every reboot.

While "free memory" is a somewhat well-defined concept on bare-metal platforms, on modern operating systems (and especially so on Linux) it is next to meaningless, so that number is to be taken with a grain of salt.

One might think, for instance, that on Linux the "correct" number (as returned by `sysinfo()`) is `struct sysinfo::freeram`, which counts completely unused memory. But when you use that on its own, you get seemingly random results.

That's because Linux uses any available memory to cache file system data, so as soon as you access any files you effectively "lose" the size of these files from `struct sysinfo::freeram`. That does not mean that that memory is unusable, though, because the cache is simply dropped as soon as the memory is needed for something else, so it's "used but available as needed". I consider that as "free" because it gives a more stable number, but that's a matter of interpretation.

The free memory count is mostly there because that's what a startup message for a traditional BASIC interpreter is supposed to look like, but don't expect it to present an accurate indication.

uli, 2024-05-29:
This should be fixed by "tscreenBase: locate(): add missing check for negative x coordinate" (which fixes the crash) and "configs/basicengine_*_lt_defconfig: enable slang, locale" in basicengine-buildroot which fixes the broken terminal output that triggered the crash.

closed 2024-05-29 by uli